garethky/klipper-voron2.4-config #2: Heat soak continuing instantly

Imburr, 2022-09-08:
First off, thanks for your brilliant work, and thanks for letting me scalp it! I put your macro into my print start macro, which I split in half. When I run it, everything fires off fine, but when it begins to soak, it triggers then instantly moves on as if the soaking is done.

Can you assist in figuring out whats going on?
[image]

Here is my start code: https://pastebin.com/3Vn68NDM

Any advice appreciated, thanks!

garethky, 2022-09-08:
so it looks like it exists here: https://github.com/garethky/klipper-voron2.4-config/blob/mainline/klipper_config/heatsoak.cfg#L123

One issue with the current implementation is that there is no smoothing of the temp sensor used for soaking. these sensors tend to have quite a bit of noise. Is it possible that the noise is causing it to calculate a low temperature rate of change? Maybe try printing the temps in the soaking phase and see?

I'm working on a patch to add the smoothing in. (Klipper smooths temp sensors attached to heaters but not the other sensors, unfortunately)

Imburr, 2022-09-08:
Hmm. Let me ask you this... With the heat soak macro should I preheat my bed? I just ran a normal print in PLA which ran heat soak, and I was interested to see what happened. It did in fact soak for longer than ABS but still not very long. I will put in the print temp as you recommend.

[image]

What do you do to exclude prints in materials that do not need to soak? Exclude in slicer?

garethky, 2022-09-08:
So the intent of the macro is that if the system has already reached thermal equilibrium it should continue immediately. So if you pre-heat your bed ahead of running the macro its going to cut down on the time the macro runs for. This saves me a bunch of time when printing. My main concern is that my build plate is thermally stable before the bed mesh is takes no its not changing shape during the meshing process. I haven't tried it on my chamber temp sensor yet because of the smoothing issue.

What you are seeing looks like noise...
```
Heat Soaking 1.080C/m / 0.75C/m
Heat Soaking 3.000C/m / 0.75C/m
Heat Soaking 4.320C/m / 0.75C/m
Heat soak completed after after ~0.2m at 27.3C
```

That is a LOT of variability in the rate of change of the chamber temp sensor. I very much doubt that your chamber temp was going up by 4.3 degrees C per minute (C/m) unless you have an active chamber heater. Basically when it quits for you is going to be random.

garethky, 2022-09-08:
Temperature smoothing is in: https://github.com/garethky/klipper-voron2.4-config/commit/f86298f2c17abb2f14f52b75ccdf59d047f56691

Try that out. You can set `SMOOTH_TIME` to change how much smoothing is applied. The smaller the rate of change you ant to detect the more smoothing you will need. Now it will soak for at least SMOOTH_TIME seconds and it wont try to evaluate the rate of temp change before its got at least that many data points. I still want to get it to continuously update but this is a start.

If that's no good I may implement @SinisterRj suggestion from here: https://klipper.discourse.group/t/interruptible-heat-soak/1552/19 to add an optional min temp check to the soaking sensor.

Imburr, 2022-09-13:
Hum, I left things as default with your time (5) and it made no difference:
[image]

I get this feeling I am doing something wrong though, and its not your code. I will see if I can print some additional detail to add insight...

garethky, 2022-09-15:
Ok, I made a lot of updates to the script. One of them is to print the quit conditions which might help you with debugging. The other is to smooth the the rate of temperature change. This requires data gathering which defaults to 20 seconds which means you should not see the macro quit instantly.

There is also a readme file that has a section explaining how to use this in PRINT_START: https://github.com/garethky/klipper-voron2.4-config/blob/mainline/klipper_config/heatsoak.readme.md

Please note that `CALLBACK` was renamed to `CONTINUE`!

Imburr, 2022-09-16:
I believe you have solved this for me! I ran it on an ABS piece last night in which I wanted the chamber to reach 36 prior to print, and it did it well. It took a while, but that's my chambers fault.

[image]

Thank you for all of your work, this is amazing. My only to do is to figure out how to only trigger the soak on certain filaments passed from the slicer, i.e.

```
gcode:
    {% set MATERIAL = params.MATERIAL|default("material") %}
    {% if MATERIAL == "ABS" %}
    PRINT_START
    {% elif MATERIAL != "ABS" %}
    PRINT_START_NOSOAK
```

garethky, 2022-09-16:
Cheers! Great that its working now. Looks like you used the fixed temperature feature.

The rate swings around quite a bit, -0.07 to + 0.3. This is where you would need to tune the smoothing for your particular sensor's noise level. You can try more temperature smoothing, more rate smoothing or both. The smoothed temps are the input to the rate calculations.

I honestly haven't used this for my chamber yet, I'm using it for my build plate (I have a sensor that's touching the magnetic sticker) so its thermally static when mesh bed leveling happens. I figure that it will take the printer ~20 minutes to get through the first 5 layers of a print anyway so by then the chamber has warmed up. My chamber takes an hour to go from 26C to 40C and I haven't seen it go higher than 55C in a 15 hour print.

Imburr, 2022-09-16:
Great detail, thank you for sharing!

closed 2022-10-03 by garethky