garethky/klipper-voron2.4-config #3: [heatsoak] PRINT_START_CONTINUE errors out but print continues

miguelangel-nubla, 2022-09-15:
When something strange happens on PRINT_START (like a homing or probing failure) the console shows an error and stops executing the rest of the gcode file (runs CANCEL macro?), but when it happens on PRINT_START_CONTINUE as described in the docs, only the rest of PRINT_START_CONTINUE is "canceled" and the gcode file resumes printing.

In my case from a klicky failure, while the probe is docked, the error cancels the rest of the current macro and any parent macro up to PRINT_START_CONTINUE, and just crashes the probe on the bed while trying to print the first layer as stated in the gcode file from which PRINT_START was originally called.

I suspect it might have something to do with the CANCEL macro not being loaded as the error action while running PRINT_START_CONTINUE? No idea about how that flow works though.

garethky, 2022-09-16:
First of all I am sorry you had a crash. Thank you for the bug report!

I'm kind of suspicious that calling a macro from another macro is different than the sd_card print process running the same macro from the gcode file. If that's the case maybe this CONTINUE functionality is unsafe and I should remove it.

Once thing you could do is to not pass the CONTINUE parameter and instead put both the PRINT_START and PRINT_START_CONTINUE in your slicer's startup gcode. This will work because HEAT_SOAK pauses the print and will only un-pause it when it succeeds. PRINT_START_CONTINUE would then be run by klipper as part of the print. If I'm right, (if!) any errors there should act to stop the print as normal. If that turns out to be safer then I can recommend that in the docs and remove the CONTINUE parameter.

I don't have the time this weekend to test as I'm not near my printer. I need a reliable way to cause an error that stops a print, like maybe commanding an illegal move beyond the X/Y max?

miguelangel-nubla, 2022-09-16:
> Once thing you could do is to not pass the CONTINUE parameter and instead put both the PRINT_START and PRINT_START_CONTINUE in your slicer's startup gcode.

That sounds like it would work as last resort, but I am sure there is a cleaner way to do this.

> I need a reliable way to cause an error that stops a print, like maybe commanding an illegal move beyond the X/Y max?

Maybe, I have been just removing the probe while probing.

---

Meanwhile I found a partial solution, moving the resume statement after the continue execution
```
            {% if heat_soak.continue %}
                {heat_soak.continue}
            {% endif %}
            {% if printer['pause_resume'].is_paused  %}
                RESUME
            {% endif %}
```
Now when a error is fired the whole macro stack gets cancelled before RESUME is ever called and the printer stays paused.
It may be even better this way than a full cancel behavior.

garethky, 2022-09-16:
Thank you for the investogation! That's is an easy enough fix to implement and it makes sense to me. Shipped: https://github.com/garethky/klipper-voron2.4-config/commit/c766a34a841216d3367281ddb167849b06e2d5bf

garethky, 2022-09-20:
Ok, turns out this has side effects: If you move the print head of change the Z offset inside the CONTINUE macro that's is reset when RESUME runs. It does wait to run RESUME until after the CONTINUE macro exists. In my case this move the print head back to the location where I had it prior to the heat soak starting.

I'm really starting to like the idea of recommending splitting the start GCode up in the slicer. I think its far less fussy vs the callback idea. For me it would look like this:
```
PRINT_WARMUP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
PRINT_START=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature] PRINT_START_X=[first_layer_print_min_0] PRINT_START_Y=[first_layer_print_min_1] PRINT_END_X=[first_layer_print_max_0] PRINT_END_Y=[first_layer_print_max_1]
```

I wanted the CONTINUE macro so I could run other macros that print afterwards, like a print test pattern, pressure advance tuning etc. I could probably find another way to get that done.

miguelangel-nubla, 2022-09-30:
What I ended up doing is to have a PRINT_START macro, position the nozzle where I want it to end, run the HEATSOAK with the continue (clean nozzle, level etc on the continue macro), then when it ends it automatically comes back to the initial position from PRINT_START.

garethky, 2022-10-03:
I haven't tried this but it looks like we could call `SAVE_GCODE_STATE NAME=PAUSE_STATE` before calling `RESUME` to overwrite the state that the `PAUSE` macro saved. You could stick it at the end of your `CONTINUE` macro. When `RESUME` runs it shouldn't move the toolhead or loose your offset.

I get the simplicity of calling 1 macro from your slicer and not 2, but that is balanced against the complexity of passing variables through callbacks in Jinja. I worked out how to do it but its error prone and difficult to debug. My view is that on balance most users would find it easier to call 2 macros.

Maybe I could leave the feature in but call it something else like "`COMPLETE`" and modify the documentation to warn not to use it for moving the print head. e.g. It could be useful for non-moving tasks, like using [Moonraker Telegram](https://github.com/Raabi91/moonraker-telegram) to let you know when to come back and watch the print start.

garethky, 2022-10-03:
> I agree. The other way around should still be documented as "advanced", just a "take care to position the nozzle where it ends at it will be reset by this macro" note. Or a list of the parameters that are being reset with the RESUME at the end.

Ok, I can work on that. Likely `CONTINUE` will be renamed to `COMPLETE` to reflect its intended purpose.

> That would assume that the PAUSE macro has not been modified by the user, for example with the popular [AndrewEllis93](https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/useful_macros.md) macros.

Agreed, I don't want to include this in my macro. You could do it in your `CONTINUE` macro so you can do the right thing for you.

> I was not aware that the offset was also modified by the default `PAUSE` and `RESUME` macros. Again, did you found a list of parameters that are being reset? Or just trial and error?

The included `PAUSE`/`RESUME` macros use `SAGE_GCODE_STATE` [here](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/pause_resume.py#L66) and [here](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/pause_resume.py#L83). (The code in AndrewEllis93's repo duplicates this which I don't totally understand???). `SAVE_GCODE_STATE` saves all of [these things](https://github.com/Klipper3d/klipper/blob/a2482d4f952e0ffd908e4381cae9e45ab1148773/klippy/extras/gcode_move.py#L210). Particularly is resets the homing position and the relative/absolute moves setting. I couldn't pick which things to reset for a user of the macro. But if you would like to exit your `CONTINUE` in the exact state that you are in at the end of that macro, calling `SAVE_GCODE_STATE NAME=PAUSE_STATE` should do that for you. It just overwrites the state that `PAUSE` saved. If you switched to relative moves for something then it might break things you you don't manually switch that back.

I'm actually going to override `RESUME` in the next update so it can shortcut the soaking phase but also stop the print from resuming if heating is still going on. You can chain overrides, so it shouldn't be an issue, you just include heatsoak.cfg after your other override.

garethky, 2022-10-06:
ok, [this change](https://github.com/garethky/klipper-voron2.4-config/commit/a1676efee780c72e9666121f9b4a202c9ae853ac) updates the documentation to reflect everything discussed here.

It also has the changed that I have been working on for Cancel and Resume buttons.

miguelangel-nubla, 2022-10-08:
Thanks for the update.

After much effort trying to keep a single PRINT_START macro I have adopted the solution to split the GCode up in the slicer. Too many workarounds are needed for the CONTINUE/COMPLETE macro to be useful with a RESUME coming after.

Until a way is found to catch errors on the CONTINUE/COMPLETE that would enable the macro to call RESUME before the CONTINUE/COMPLETE macro is called and not risk the behavior mentioned in this issue, splitting the macro in the slicer will be the best way to go.

closed 2022-10-08 by miguelangel-nubla