garethky/klipper-voron2.4-config #5: LCD menu Z-Offset doesn't save to sheet manager

Ibot-11, 2022-10-28:
I found an issue with the default LCD display of the Voron.
The default menu.cfg uses Z_Offset instead of Z_Ajust. Therefore the sheet manager script doesn't save the Offset changes.
Here is the part of the default menu.cfg.

```
[menu __main __tune __offsetz]
type: input
name: Offset Z:{'%05.3f' % menu.input}
input: {printer.gcode_move.homing_origin.z}
input_min: -5
input_max: 5
input_step: 0.005
realtime: True
gcode:
   SET_GCODE_OFFSET Z={'%.3f' % menu.input} MOVE=1
```
I don't get it to work with Z_Adjust. Could you help me with this one?
Thank you.

garethky, 2022-10-28:
Ahh crud, [the default](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/display/menu.cfg#L122) is to specify an absolute value...

I don't have the default LCD hooked up to my Voron anymore to verify that this works, but I think you want to compute the difference between the menu input value and the current offset and then apply it as an adjustment:

```
gcode:
    SET_GCODE_OFFSET Z_ADJUST={'%.3f' % (menu.input - printer.gcode_move.homing_origin.z)} MOVE=1
```

Can you please let me know if that's a fix so I can update the documentation?

Ibot-11, 2022-10-28:
I tried the changed gcode you sent me. Works like a charm!
Thank you!

I also found another problem I can’t solve myself.
When i use SHOW_BUILD_SHEETS I only get this error:
`Error evaluating 'gcode_macro SHOW_BUILD_SHEETS:gcode': ValueError: too many values to unpack.`

My variables.cfg:
```
[Variables]
build_sheet flat = {'name': u'flat', 'offset': 0.0}
build_sheet installed = 'build_sheet textured'
build_sheet textured = {'name': u'textured', 'offset': -0.07000000000000002}
```

Should I open a new issue for this to keep it separate from this one?

garethky, 2022-10-29:
Ok, interesting, I don't think I have seen that before. My file looks very similar to yours:

```
[Variables]
build_sheet installed = 'build_sheet smooth_pei'
build_sheet smooth_pei = {'name': 'Smooth PEI', 'offset': -0.06499999999999997}
build_sheet textured_pei = {'name': 'Textured PEI', 'offset': -0.14}
```

The only thing I see that's different is the 'u' before the names of your sheets. This is probably [something to do with your version of Python](https://stackoverflow.com/questions/2464959/whats-the-u-prefix-in-a-python-string).

```
pi@voron24:~/klipper_config $ python --version
Python 2.7.16
```

Whats version of Python are you running on your printer? 3.3+ ?

Ibot-11, 2022-10-29:
My printer is also running 2.7.16.
But there is also a Python3 when I do `pip3 –version`
I don't know if this is any relevant to the issue.

```
pi@lightblue:~/klipper_config $ python --version
Python 2.7.16
pi@lightblue:~/klipper_config $ pip3 --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
pi@lightblue:~/klipper_config $
```
I also tried to remove the .4
The variables work fine without it, but still the same error when using SHOW_BUILD_SHEETS

garethky, 2022-10-31:
Ok, I was able to reproduce this and fix it. This is a hisenbug, now that I see it I have no idea how this code ever worked!

Ibot-11, 2022-11-04:
Thank you! Works without any problems now.

closed 2022-11-04 by Ibot-11