garethky/klipper-voron2.4-config #7: Help With What is Wrong

Imburr, 2022-11-15:
Garethky,

Thank you again for your great work on this! I am having a few issues, perhaps you can help pinpoint what I am doing wrong. I have a version of your macro which is set to a 10m timeout. Basically, I want the plate to soak for 10m and then start printing. But when it reaches the timeout, it just sits there. If I hit "RESUME" it then starts to immediately print without doing the PRINT_START_CONTINUE macro. If I manually print start continue, the temps are messed up (defaulted).

What needs to happen for the print to continue once it reaches a 10m timer? Here is a screenshot, you can see where I was trying to force the macros to continue: https://i.imgur.com/s5jpdpu.png

I also have this issue with my main soak macro, sometimes it does not reach the smooth, and it times out, and then it just sits there indefinitely at my pre-mesh temps.

This is not an issue with your code- if there is a better place for me to put this such as a DM or discord or reddit, please let me know!

garethky, 2022-11-15:
Ok, that's a lot to untangle. Not saying I can solve your use-case but I appreciate the feedback and understand your confusion/frustration. Let me see if I understand:

#### Timeout
The timeout feature only works in the Soaking phase. I'm going to make an update to the docs to reflect this better. If you want a timeout in the heating phase, that isn't supported. The assumption is the temp sensor for the heating phase is an actively heated and PID controlled sensor that can hit the target temp and stay there. So there isn't any need for a timeout in that phase.

Maybe I should make a safety check on the supplied temp sensor to make sure its attached to a heater so this isn't done incorrectly.

The timeout feature is a safety feature intended to end the print. Its not supposed to print after that. It should be running the cancel script when the timeout occurs. I think I need to make sure that happens in the code.

#### Cancel/Resume

The image you shared I think I understand... The timeout was triggered but there is no `STOP_HEAT_SOAK` in your cancel macro, so it never stops. 😢 This is a sharp edge and ideally I wouldn't have to require that you do that.

Just generally, the complexity around `CANCEL`/`RESUME` is not what I had hoped for and should be improved. It works how I want it to work once its all configured but its too many things for people to do correctly all at once. It needs to degrade gracefully and be safer out of the box with no mods.

The way overrides work, the last one wins, you cant chain them. This a Klipper limitation. So you have to do everything in 1 macro.

#### Changes
- Check that the Heater is really a Heater and not a temp sensor
- If there is a print running, (virtual_sdcard is printing) call `CANCEL_PRINT` as the default behavior for a timeout. I think for most people that means you wont have to pass a cancel macro.
- Call `STOP_HEAT_SOAK` whenever the timer macro stops. Never assume the user is going to do it in CANCEL_PRINT or whatever. (I think this fixes your screenshot)
- If there is a print running when the macro starts and that changes, assume cancel was run, run `STOP_HEAT_SOAK` and abort. This removes a sharp edge around CANCEL, if its not provided or it doesn't contain `STOP_HEAT_SOAK` you will still get reasonable behavior. An error message can explain what to do to your config to get the error to go away.
- If there is a print running and the timer macro finds that the print is not in the PAUSE state, run `STOP_HEAT_SOAK` and abort. An error message can explain what to do to your config to get the error to go away.
- Split out a `HEAT_SOAK_RESUME` macro that you can use from your own RESUME override to get the nice resume behaviour.
- Update the docs to include the macro before your own overrides of `CANCEL_PRINT` and `RESUME` not after.

I have like 2 of these things done already but they need testing.

Imburr, 2022-11-16:
I am the one introducing complexity here, and for that I am sorry. I really have two use cases for your heat soak. One of them works 90% of the time, the other I am only beginning to work on.

Case 1:

ABS, chamber door closed: I run a heat soak with this call:

HEAT_SOAK SOAKER='temperature_sensor chamber_temps' SOAK_TEMP=35 RATE=0.2 RATE_SMOOTH=30 CONTINUE='{CONTINUE}'

Once the chamber thermistor reaches 35 degrees (which takes 10-15 minutes), the print starts. This is done simply to prevent me from beginning to print with ABS while the chamber is too cool. This works great most of the time.

Case 2:

PLA, chamber door open: I want to heat soak the bed at 60 degrees for 10 minutes, then perform my mesh just to make sure this is evenly heated. In this case I am not concerned about the chamber thermistor/temp, and I just want the bed to chill at 60 for 10m.

I have a sinking suspicion my START_PRINT and START_PRINT_CONTINUE are funky. Let me ask you this, do I need to pass variables from START_PRINT to continue? Am I losing my temps in the midst of the handoff?

START_PRINT (Just top portion):

```
[gcode_macro START_PRINT]
variable_parameter_EXTRUDER_TEMP : 0
variable_parameter_BED_TEMP : 0
variable_callback : None
gcode:
  {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(190)|float %}
  {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
  {% set CALLBACK = params.CALLBACK %}
  SET_GCODE_VARIABLE MACRO=START_PRINT VARIABLE=callback VALUE="'{CALLBACK}'"
  {% set CONTINUE -%}
       PRINT_START_CONTINUE EXTRUDER_TEMP={EXTRUDER_TEMP} BED_TEMP={BED_TEMP}
  {%- endset %}
```

START_PRINT_CONTINUE:

```
[gcode_macro PRINT_START_CONTINUE]
variable_parameter_EXTRUDER_TEMP : 0
variable_parameter_BED_TEMP : 0
gcode:
  {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(205)|float %}
  {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
```

If this is beyond the scope of your ability to help, no worries just trying to figure this up!

garethky, 2022-11-16:
For case #2 you can install a "top bed" temp sensor. I used one of the spare mounting holes in the build plate to install mine. It doesn't have to be in the middle of the plate to work well. Mine is at the edge. Not having to wait those 10 minutes every time was why I worked on the macro. If you really want to wait for 10 minutes just do a dwell:

```
G4 P600000
```

The `START_PRINT` stuff looks correct to me but I stopped using it myself because it was too tricky to get right all the time. If you suspect you have an issue try writing the variables to the console in `PRINT_START_CONTINUE`. That should shed some light on whats happening. Myself, I just call both macros from my slicer now.

Imburr, 2022-11-16:
When do you call CONTINUE from slicer?

Imburr, 2022-11-16:
Can you extrapolate on the top bed sensor? I assume its just a thermistor under the build plate, which reads a secondary temp? If you put it towards the outside, I suppose you could just tell the macro to heat soak till its stable at 60.

garethky, 2022-11-16:
> When do you call CONTINUE from slicer?

I don't, the script pauses the print and resumes it for you. See the documentation for the recommended setup: https://github.com/garethky/klipper-voron2.4-config/blob/mainline/klipper_config/heatsoak.readme.md#heat-soaking-as-part-of-print_start

My slicer has something similar to this:
```
PRINT_WARMUP EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
PRINT_START EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
```

> Can you extrapolate on the top bed sensor?

Just a glass bead thermistor that's shoved up to touch the magnetic sheet and secured with some kapton tape.

> I suppose you could just tell the macro to heat soak till its stable at 60.

If your heater is set to 60C you will never see 60C at the top of the bed due to thermal loss. The exact temp you will get vary depending on the build sheet, if the enclosure is open or closed, ambient temp etc. The whole point of this is you don't want to guess what that temperature is, all you care about is that its **_stable_**. This script calculates temperature rate of change and waits only as long as is needed to get to stability. I just do [this](https://github.com/garethky/klipper-voron2.4-config/blob/daa503991c7a65bc5fe33102ba32d4ee4a581fad/klipper_config/macros.cfg#L171):

```
    # wait for top bed sensor to be in thermal equilibrium:
    HEAT_SOAK HEATER='heater_bed' TARGET={BED_TEMP} SOAKER='temperature_sensor top_bed' CANCEL='CANCEL_PRINT'
```

Imburr, 2022-11-18:
When you say "Just a glass bead thermistor that's shoved up to touch the magnetic sheet and secured with some kapton tape.", can you elaborate? I have an ender aluminum build plate what has heating element on the bottom, Then on top of the build plate I have a magnetic sticker (which I can remove if needed, I have another. Then on top of that spring steel.

Did you carve out a spot in the magnet so the bead doesnt bulge the magnet up?

garethky, 2022-11-21:
If you have the kind of bed where there is a magnetic sheet on top, drill a hole near the edge and put the thermistor though the hole until it touches the underside of the magnetic sheet. I had a pre-drilled hole that I used. The magnetic sheet is not disturbed.
[image]

Another way to do it is to cut a slot in the top of the bed and install the thermistor there with some high temp silicone adhesive (such a automotive gasket sealant). I cut this slot with a metal cutting circular saw blade. Its only about 4mm deep:

[image]

closed 2022-11-21 by Imburr

Imburr, 2022-11-24:
Your awesome thanks. Modifying my bed, changing some sensors around- ultimate goal is what you have with a top_bed sensor.

I also changed out my chamber thermistor for a BME280, and it correctly handles the rate:

```
3:29 PM
Soaking -- temp: 36.8C / 35.0C -- rate: 0.225C/m / 0.200C/m -- 18.2m remaining
3:29 PM
Soaking -- temp: 36.8C / 35.0C -- rate: 0.218C/m / 0.200C/m -- 18.1m remaining
3:29 PM
Soaking -- temp: 36.8C / 35.0C -- rate: 0.218C/m / 0.200C/m -- 18.1m remaining
3:29 PM
Heat soak complete after ~12.0m at 0.198C/m / 36.8C
```

Thanks again!

Imburr, 2022-12-02:
One additional question- the second picture, what is the gray material holding in the thermistor cable? I have high temp silicone, and I also have thermal grease. I was going to encase the thermistor in thermal grease, but I was worried about it "getting everywhere" and i pop the bed on and off.

garethky, 2022-12-02:
It's High Temp Silicone, specifically Hondabond HT for exhaust gaskets Hondabond High-Temp Silicone Liquid Gasket 1.9 fl oz https://a.co/d/4ThauBV

But I think any high temp silicone would work.