garethky/klipper-voron2.4-config #9: Variables do not appear to be used when manually set in cfg file and must be called

tastyratz, 2023-11-10:
I have been spending a lot of time trying to just configure all of my variables in heatsoak.cfg so I can just click the macro HEAT_SOAK and let it run. Unfortunately, it seems like values populated there do not in fact get called from.

If I attempt to just run the macro I get the below error:

```
1:29 PM
Error evaluating 'gcode_macro HEAT_SOAK:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute ''
1:29 PM
Heat Soak starting
1:29 PM
HEAT_SOAK
```

If I attempt to populate a value and not call it out in the command, for example, heatsoak just doesn't run (nor give any indication it isn't running)

I.E. if I run:

`HEAT_SOAK HEATER='heater_bed' TARGET=110 SOAKER='temperature_sensor Chamber' SOAK_TEMP=40 RATE=0.3 TEMP_SMOOTH=4 RATE_SMOOTH=20 COMPLETE='SOAKING_COMPLETE' CANCEL='CANCEL_PRINT' TIMEOUT=90 HEATING_REPORT_INTERVAL=30 SOAKING_REPORT_INTERVAL=60`

Everything works great!

If I take something out like Target=110 and populate it in the configuration it tells me Heat Soak Starting, has no error messages, and doesn't do anything.

Here is my variable list in heatsoak.cfg:

```
[gcode_macro HEAT_SOAK]
description: Wait for a temperature sensor to reach thermal equilibrium

variable_last_temp: 0
variable_last_soak_temp_rate: None
variable_temp_history: None
variable_smoothed_temp_history: None
variable_stage: 'done' ## heating -> soaking -> done -> None
variable_total_time_elapsed: 0
variable_soak_time_remaining: 30
variable_check_interval: 300
variable_heating_report_interval: 30
variable_soaking_report_interval: 60
variable_heater_sensor: "heater_bed"
variable_soaker_sensor: "temperature_sensor Chamber"
variable_target_temp: 110.0
variable_min_soak_temp: 35
variable_target_rate: 0.3
variable_complete: None
variable_cancel: None
variable_temp_smooth_time: 4.0
variable_rate_smooth_time: 20.0
variable_resume_trigger: False
variable_was_print_active: False

```

garethky, 2023-11-10:
Sorry you are having trouble. I would suggest just writing your own wrapper macro that calls this macro. Use [Macro Parameters](https://www.klipper3d.org/Command_Templates.html#macro-parameters) to pass in what you want to use and hard code whatever you want as constant.

Understanding all the ins and out of the way klipper handles these macros is hard, bordering on unusable. This script is really at the outer limit of what anyone should try to do without resorting to Python. You have to grock [Macro Parameters](https://www.klipper3d.org/Command_Templates.html#macro-parameters), [Variables](https://www.klipper3d.org/Command_Templates.html#variables) and [Delayed Gcodes](https://www.klipper3d.org/Command_Templates.html#delayed-gcodes). I don't think the docs do a good job of explaining it all and I did a lot of experimenting to get it all to work. Best to just treat this like its a built-in GCode and not touch it.

ryukumon, 2024-01-15:
> I have been spending a lot of time trying to just configure all of my variables in heatsoak.cfg so I can just click the macro HEAT_SOAK and let it run. Unfortunately, it seems like values populated there do not in fact get called from.
> 
> If I attempt to just run the macro I get the below error:
> 
> ```
> 1:29 PM
> Error evaluating 'gcode_macro HEAT_SOAK:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute ''
> 1:29 PM
> Heat Soak starting
> 1:29 PM
> HEAT_SOAK
> ```
> 
> If I attempt to populate a value and not call it out in the command, for example, heatsoak just doesn't run (nor give any indication it isn't running)
> 
> I.E. if I run:
> 
> `HEAT_SOAK HEATER='heater_bed' TARGET=110 SOAKER='temperature_sensor Chamber' SOAK_TEMP=40 RATE=0.3 TEMP_SMOOTH=4 RATE_SMOOTH=20 COMPLETE='SOAKING_COMPLETE' CANCEL='CANCEL_PRINT' TIMEOUT=90 HEATING_REPORT_INTERVAL=30 SOAKING_REPORT_INTERVAL=60`
> 
> Everything works great!
> 
> If I take something out like Target=110 and populate it in the configuration it tells me Heat Soak Starting, has no error messages, and doesn't do anything.
> 
> Here is my variable list in heatsoak.cfg:
> 
> ```
> [gcode_macro HEAT_SOAK]
> description: Wait for a temperature sensor to reach thermal equilibrium
> 
> variable_last_temp: 0
> variable_last_soak_temp_rate: None
> variable_temp_history: None
> variable_smoothed_temp_history: None
> variable_stage: 'done' ## heating -> soaking -> done -> None
> variable_total_time_elapsed: 0
> variable_soak_time_remaining: 30
> variable_check_interval: 300
> variable_heating_report_interval: 30
> variable_soaking_report_interval: 60
> variable_heater_sensor: "heater_bed"
> variable_soaker_sensor: "temperature_sensor Chamber"
> variable_target_temp: 110.0
> variable_min_soak_temp: 35
> variable_target_rate: 0.3
> variable_complete: None
> variable_cancel: None
> variable_temp_smooth_time: 4.0
> variable_rate_smooth_time: 20.0
> variable_resume_trigger: False
> variable_was_print_active: False
> ```
> 
> Ideally, I want to create a button in mainsail I can click in the menu that gets a heatsoak running without typing so I can start it from my phone in the shop or without thinking.

I haven't looked at the coding close enough to come up with the right solution BUT you can add a default value inline with each variable like in the picture to accomplish what you want. Still reacts to parameters set from the console properly as well.

[image]

This is what I did for the time being.

ryukumon, 2024-01-16:
> I have been spending a lot of time trying to just configure all of my variables in heatsoak.cfg so I can just click the macro HEAT_SOAK and let it run. Unfortunately, it seems like values populated [...]

Ok I looked at the code more closely and you can copy this block below and replace it in the heatsoak.cfg and it will set the defaults according to the variables within the macro:

`
    # SOAKER is no longer required IF you set soaker_sensor in the variables above
    {% set SOAKER = params.SOAKER | default(printer['gcode_macro HEAT_SOAK'].soaker_sensor) | string %} # | default('temperature_sensor Chamber') 
    {% set RATE = params.RATE | default(printer['gcode_macro HEAT_SOAK'].target_rate) | float %} ## in degrees C per minute
    {% set SOAK_TEMP = params.SOAK_TEMP | default(printer['gcode_macro HEAT_SOAK'].min_soak_temp) | float %}
    {% set HEATER = (params.HEATER | default(printer['gcode_macro HEAT_SOAK'].heater_sensor) | string) %} # Optional heater | default('heater_bed') 
    {% set TARGET = params.TARGET | default(printer['gcode_macro HEAT_SOAK'].target_temp) | float %}
    {% set TIMEOUT = (params.TIMEOUT | default(printer['gcode_macro HEAT_SOAK'].soak_time_remaining) | int) * 60 %} ## minutes to seconds
    {% set TEMP_SMOOTH = params.TEMP_SMOOTH | default(printer['gcode_macro HEAT_SOAK'].temp_smooth_time) | float %} ## seconds for temp smoothing
    {% set RATE_SMOOTH = params.RATE_SMOOTH | default(printer['gcode_macro HEAT_SOAK'].rate_smooth_time) | float %} ## seconds for rate smoothing
    {% set COMPLETE  = (params.COMPLETE | string) %}
    {% set CANCEL  = (params.CANCEL | string) %}
    {% set HEATING_REPORT_INTERVAL  = params.HEATING_REPORT_INTERVAL | default(printer['gcode_macro HEAT_SOAK'].heating_report_interval) | int %}
    {% set SOAKING_REPORT_INTERVAL  = params.SOAKING_REPORT_INTERVAL | default(printer['gcode_macro HEAT_SOAK'].soaking_report_interval) | int %}
`

ryukumon, 2024-01-16:
> I have been spending a lot of time trying to just configure all of my variables in heatsoak.cfg so I can just click the macro HEAT_SOAK and let it run. Unfortunately, it seems like values populated [...]

Ok, I looked at the code more closely and changed it to the code below. You should be able to copy this code block and paste it over the corresponding block in the heatsoak.cfg and it will set defaults to the variables within the macro:

```
    # SOAKER is no longer required IF you set soaker_sensor in the variables above
    {% set SOAKER = params.SOAKER | default(printer['gcode_macro HEAT_SOAK'].soaker_sensor) | string %} # | default('temperature_sensor Chamber') 
    {% set RATE = params.RATE | default(printer['gcode_macro HEAT_SOAK'].target_rate) | float %} ## in degrees C per minute
    {% set SOAK_TEMP = params.SOAK_TEMP | default(printer['gcode_macro HEAT_SOAK'].min_soak_temp) | float %}
    {% set HEATER = (params.HEATER | default(printer['gcode_macro HEAT_SOAK'].heater_sensor) | string) %} # Optional heater | default('heater_bed') 
    {% set TARGET = params.TARGET | default(printer['gcode_macro HEAT_SOAK'].target_temp) | float %}
    {% set TIMEOUT = (params.TIMEOUT | default(printer['gcode_macro HEAT_SOAK'].soak_time_remaining) | int) * 60 %} ## minutes to seconds
    {% set TEMP_SMOOTH = params.TEMP_SMOOTH | default(printer['gcode_macro HEAT_SOAK'].temp_smooth_time) | float %} ## seconds for temp smoothing
    {% set RATE_SMOOTH = params.RATE_SMOOTH | default(printer['gcode_macro HEAT_SOAK'].rate_smooth_time) | float %} ## seconds for rate smoothing
    {% set COMPLETE  = (params.COMPLETE | string) %}
    {% set CANCEL  = (params.CANCEL | string) %}
    {% set HEATING_REPORT_INTERVAL  = params.HEATING_REPORT_INTERVAL | default(printer['gcode_macro HEAT_SOAK'].heating_report_interval) | int %}
    {% set SOAKING_REPORT_INTERVAL  = params.SOAKING_REPORT_INTERVAL | default(printer['gcode_macro HEAT_SOAK'].soaking_report_interval) | int %}
```