garethky/klipper-voron2.4-config #10: Heatsoak macro is not pausing on Octoprint

SinisterRj, 2023-11-16:
Hi Gareth,

The `{% set is_print_active = printer['virtual_sdcard'].is_active or printer['virtual_sdcard'].file_position != 0.0 %}` line on heatsok,cfg is not abble to set if the printer is running or not when using Octoprint instead Mainsaill.

garethky, 2023-11-16:
So as you can probably tell from that line I am having to _guess_ if a print is running. I don't use Octoprint so I cant easily investigate this myself.

Questions I have are:
* What is Octoprint doing to run a print that doesn't use the `virtual_sdcard`?
* How should we know that Octoprint is running a print? Is there some other object that we should look at?

SinisterRj, 2023-12-30:
This is what worked for me:

{% set is_print_active = printer['virtual_sdcard'].is_active or printer['virtual_sdcard'].file_position != 0.0 **or printer.idle_timeout.state == 'Printing'** %}

And I really don't know what Octoprint does.. Lol.

P.S.: Sorry for the delay.

closed 2023-12-30 by SinisterRj

garethky, 2023-12-30:
Oh thats cool! I guess I need to go back and see if this needs checking anywhere else that I look at the state of the `virtual_sdcard`. Maybe this is better test for "is printing" in general.